austinbv/dino #81: Program stops on board att

jotapem, 2015-01-14:
I installed Dino yesterday and I'm having problems trying to execute the simple blinking program.
I realized that the program 'gets stuck' on the `board = Dino::Board.new(Dino::TxRx::Serial.new)`line. 
(I also tried  `board = Dino::Board.new(Dino::TxRx.new)`)

I kept regenerating and reuploading the sketches and a few minutes ago I got the Board Not Found error. Then I reinstalled Dino and I'm back to the same situation: the program is running forever and is not logging a thing.

vickash, 2015-01-14:
Which arduino board and what OS?

jotapem, 2015-01-14:
arduino UNO and windows 7

vickash, 2015-01-14:
Which branch? master simply won't work at all on Windows because of a problem with the `serialport` gem. You'd need to try 0.12, which I switched to use the `rubyserial` gem a few days ago. Have not tested on Windows yet, but it should work.

jotapem, 2015-01-14:
I replaced the files with the 0.12 branch files, but now I'm getting these errors on the `require 'dino'` line

[image]

vickash, 2015-01-14:
Run `bundle install`

jotapem, 2015-01-14:
the bundle is now complete and the `require 'dino'` line does not log any errors, but the `board = Dino::Board.new(Dino::TxRx.new)` now log these:

[image]

vickash, 2015-01-14:
Ruby's still finding the old version (master branch), you installed via rubygems. 0.11.2 and 0.12.0 are not compatible. Remove the old one by running `gem uninstall dino`. Remove everything.

You can install 0.12 into your system by building it and then installing. In the source directory for 0.12 run:
`gem build dino.gemspec`
`gem install dino-0.12.0.gem`

jotapem, 2015-01-22:
I succesfully ran the 3 lines you told me to, but I'm getting errors on `dino` and `dino generate-sketch serial`

[image]

I figured I should have deleted even more stuff, so I just reinstalled Ruby then built dino-0.12 and I'm getting the exact same error log

vickash, 2015-02-01:
Check if file `C:/Ruby193/lib/ruby/gems/1.9.1/gems/dino-0.12.0/lib/dino_cli.rb` exists. It's saying it can't find the file that should be at that path.

I tried building the gem from source in Ruby 1.9.3 under windows XP and couldn't recreate your error.

closed 2018-02-17 by vickash