austinbv/dino #84: For Newbies

hanjeongjin, 2015-02-16:
Installed dine, generated sketch i.e. du.ino
But no blinking.
What's wrong?

Using ruby200, arduino  r3

vickash, 2015-02-16:
What OS?

hanjeongjin, 2015-02-19:
thank you for the reply Sir, I am using Windows 7 Starter 32 bit OS

vickash, 2015-02-19:
Sorry. Master branch and rubygem versions don't work at all on Windows. See issue #81 for an explanation on how to get the latest branch (0.12), which works fine on Windows. You will need to uninstall the old version of the gem and reflash the board with the firmware included in 0.12.

vickash, 2015-02-19:
Also, I've changed the command line syntax slightly. It should now be `dino sketch serial`, to create the sketch folder, which is simpler.

closed 2018-02-17 by vickash