austinbv/dino #86: Latest .ino file not verified: Invalid initialization of reference of type 'HardwareSerial&'

sscirrus, 2015-05-15:
Hi there,

I just installed dino on a new app, ran `dino sketch serial` and opened the `dino_serial.ino` in the Arduino IDE. It is failing to compile: 'Invalid initialization of reference of type 'HardwareSerial&' from expression of type 'usb_serial_class'.

The full error message stack:

```
var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp: In member function 'void DHT::readSensor()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp:165:17: warning: 'data' may be used uninitialized in this function [-Wmaybe-uninitialized]
data <<= 1;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp:147:8: warning: 'rawTemperature' may be used uninitialized in this function [-Wmaybe-uninitialized]
word rawTemperature;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp:198:28: warning: 'rawHumidity' may be used uninitialized in this function [-Wmaybe-uninitialized]
humidity = rawHumidity * 0.1;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp: In member function 'void Dino::writeResponse()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp:120:22: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
_writeCallback("\n");
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp: In member function 'void Dino::ds18Read()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp:355:8: warning: unused variable 'present' [-Wunused-variable]
byte present = ds.reset();
^
dino_serial:21: error: invalid initialization of reference of type 'HardwareSerial&' from expression of type 'usb_serial_class'
invalid initialization of reference of type 'HardwareSerial&' from expression of type 'usb_serial_class'
```

Environment:
Rails 4.2.1
Ruby 2.0.0p645
Teensy 3.1 w/Teensyduino 1.22
Arduino 1.6.3

Note that sketches I've created in Arduino IDE itself do work properly when uploaded to the device.

vickash, 2015-05-15:
The Teensy uses an ARM chip, but not one that is identical to the chip used in any official Arduino board (which is all the hardware I've targeted so far). There is an official Arduino with a ARM, the Due. I've had to make some changes and use preprocessor directives to get even that working.

That said, it works, and it should work on the Teensy with some tweaking.

First, the error you're getting has to do with the directives that "choose" the right serial object and create a pointer to it.

That's this bit of code:
https://github.com/austinbv/dino/blob/0.12.0-wip/src/dino_serial.ino#L14-L22

Depending on the architecture, and specific model even, a different line of code gets run to determine which serial port dino should use and create a pointer in the `serial` variable.

I did it this way so that, if you want to use a different interface on a board with multiple UARTS, it just needs to be changed in one place. Rather than trying to find and replace all calls to `Serial` with `Serial3` or whatever.

Anyway, yours is probably falling into the "else" branch. Based on the code on the Teensy site, you want `Serial_ &serial = Serial;` i.e. same thing as the 32U4 boards (Leonardo family).

You could just blow away all that code, and replace it with that one line. Or you can try modifying the line for the 32U4 to add support for the Teensy.
````
#elif defined(__AVR_ATmega32U4__) || defined(__MK20DX128__) || defined(__MK20DX256__)
  Serial_ &serial = Serial;
````
The two strings starting with __MK  should test for the Teensy 3.01 and 3.1 respectively, causing it to assign the serial object properly.

If you do that and get another error, I'm listing below the other places where I used directives to cater specifically to the Due (ARM) Arduino board.

dino_serial.ino:
https://github.com/austinbv/dino/blob/0.12.0-wip/src/dino_serial.ino#L8-L10 (no soft serial support for Due last time I checked. Other ARMs may have the same problem)

Dino.cpp:
https://github.com/austinbv/dino/blob/0.12.0-wip/src/lib/Dino.cpp#L7-L9 (C lib for double to ascii. Due bug?)
https://github.com/austinbv/dino/blob/0.12.0-wip/src/lib/Dino.cpp#L14-L17 (omit soft serial again)

You may need to modify these directives also to match the Teensy boards (not just the Due). If you do end up fixing everything this way, not altering compatibility with the Arduino boards, I'll gladly accept a pull request.

I don't have a Teensy, but let me know if you run into any other bugs.

sscirrus, 2015-05-15:
Thank you Vickash!

First, the code you suggested (adding MK20DX128 and MK20DX256) did fix the first error.

There is one left, which is: `serial was not declared in this scope` pointing to the `void loop()` block. I believe that once this one is fixed, Teensy will be working!

Full trace:
```
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp: In member function 'void DHT::readSensor()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp:165:17: warning: 'data' may be used uninitialized in this function [-Wmaybe-uninitialized]
data <<= 1;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp:147:8: warning: 'rawTemperature' may be used uninitialized in this function [-Wmaybe-uninitialized]
word rawTemperature;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/DHT.cpp:198:28: warning: 'rawHumidity' may be used uninitialized in this function [-Wmaybe-uninitialized]
humidity = rawHumidity * 0.1;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp: In member function 'void Dino::writeResponse()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp:120:22: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
_writeCallback("\n");
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp: In member function 'void Dino::ds18Read()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build6615047894839573031.tmp/Dino.cpp:355:8: warning: unused variable 'present' [-Wunused-variable]
byte present = ds.reset();
^
dino_serial:19: error: 'Serial_' does not name a type
dino_serial.ino: In function 'void writeResponse(char*)':
dino_serial:25: error: 'serial' was not declared in this scope
dino_serial.ino: In function 'void setup()':
dino_serial:29: error: 'serial' was not declared in this scope
dino_serial.ino: In function 'void loop()':
dino_serial:40: error: 'serial' was not declared in this scope
'serial' was not declared in this scope
```

vickash, 2015-05-15:
I made an error. The answer was in your first message actually. Forget what I said about adding conditions to the 32U4 branch (the other ARM patches still apply).

The `Serial` object is not of the `Serial_` class in the Teensy like the 32U4. Apparently, the class name is `usb_serial_class`? Or so your backtrace says. Let's see if it works.

Try adding this branch to the ifdef instead:
````
#elif defined(__AVR_ATmega32U4__) || defined(__MK20DX128__) || defined(__MK20DX256__)
  usb_serial_class &serial = Serial;
````

vickash, 2015-05-15:
That should actually be:
````
#elif  defined(__MK20DX128__) || defined(__MK20DX256__)
  usb_serial_class &serial = Serial;
````
Forgot the 32U4 in there. You don't want to match that.

sscirrus, 2015-05-21:
Vickash,

It seems to verify and upload now - thank you. I don't know if these are significant - I am getting these messages highlighted in red when verifying/uploading:

```
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/DHT.cpp: In member function 'void DHT::readSensor()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/DHT.cpp:165:17: warning: 'data' may be used uninitialized in this function [-Wmaybe-uninitialized]
data <<= 1;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/DHT.cpp:147:8: warning: 'rawTemperature' may be used uninitialized in this function [-Wmaybe-uninitialized]
word rawTemperature;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/DHT.cpp:198:28: warning: 'rawHumidity' may be used uninitialized in this function [-Wmaybe-uninitialized]
humidity = rawHumidity * 0.1;
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/Dino.cpp: In member function 'void Dino::writeResponse()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/Dino.cpp:120:22: warning: deprecated conversion from string constant to 'char*' [-Wwrite-strings]
_writeCallback("\n");
^
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/Dino.cpp: In member function 'void Dino::ds18Read()':
/var/folders/rd/24281_h15x5dxnd5c__z25jh0000gp/T/build8867303613332974087.tmp/Dino.cpp:355:8: warning: unused variable 'present' [-Wunused-variable]
byte present = ds.reset();
^

Sketch uses 44,248 bytes (16%) of program storage space. Maximum is 262,144 bytes.
Global variables use 7,124 bytes (10%) of dynamic memory, leaving 58,412 bytes for local variables. Maximum is 65,536 bytes.
```

vickash, 2015-05-23:
Glad you got it to compile. Not 100% sure about those messages, but the first 3 definitely have something to do with the DHT11 / DHT 21 / DHT 22 sensor library.

Have you tried using the board after uploading? Does it work? If you have one of those sensors you can try it and see if it works. Those might be nothing to worry about.

vickash, 2018-01-14:
The original issue here, the type of `Serial` not being define correctly, should no longer be a problem as long as the Arduino board definition defines any constant with the name `Serial`, thanks to [these changes](https://github.com/austinbv/dino/pull/34/commits/78fb692bbbd8b2914f85fc631d3fd932549e0076#diff-1bb578ed4865ea8e9c9310c2a2960a8cL5), which use `#define` to set `serial` as a reference to `Serial`, rather `if` which requires explicit type. The compiler just takes care of it.

closed 2018-01-14 by vickash