austinbv/dino #91: Should I keep the connection ?

MwMaki, 2020-08-07:
Hi, This is a question more than an issue.
I'm new in using the UNO with ruby, I did write code and it's working but when I disconnect the uno I lose everything!

Do I need to keep the connection between the Uno and the computer?
How can I upload the ruby code, disconnect the uno and still have the code running on the uno?

vickash, 2020-08-12:
Yes, you need to maintain the connection to the board.

The board is flashed with an Arduino sketch that waits for commands and processes them, nothing else. The Ruby script you write runs on your computer, and sends commands to the board. It needs to keep running on your computer, with a persistent connection to the board. Ruby code is never sent to or interpreted by the board.

If you really need to write in a scripting language, then have that code run on a standalone microcontroller, I suggest checking out [micropython](https://micropython.org/). Not aware of any Ruby equivalent.

Hope that helps.

MwMaki, 2020-08-12:
@vickash  Thanks you for your response, It's helpful and did answer my questions and concerns

closed 2020-08-17 by MwMaki